qingfengxia/Cfd #11: Pressure unit

Worufu3D, 2017-05-23:
Pressure unit is displayed as mm*kg/s^2 in the various constraint form (ambient initial pressure, outlet pressure, inlet, etc...), that is equal to mN (milliNewton). Not the right pressure unit, however I think it is better to use Pa as pressure unit, more used than kg*m^-1*s^-2.

Worufu3D, 2017-05-23:
When I try to insert velocity or mass flow in boundary condition, the form doesn't accept values under 1 m/s or kg/s.
If I try to put 3 mm/s, after close boundary form (with OK button), when I reopen the form, the value is displayed as 3000 mm/s.

qingfengxia, 2020-01-22:
The C++ class TaskPanel does not support Expression yet, is that from CfdOF fork?  Cfd currently fix unit to SI unit.

closed 2020-01-22 by qingfengxia