qingfengxia/Cfd #13: Input values return wrong storage

Worufu3D, 2017-06-15:
I found some errors as the old, and fixed, inlet velocity multiplication by 1000, or other factor after parameter tab closure. These errors are present in various boundary and field form value. For example, I tried the new tutorial, but I can not start it beacuse the value of turbolence intensity (set as 5) goes out of range (stored as 50).

[image]

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.11260 (Git)
Build type: Release
Branch: master
Hash: 086d2ed0bbe75916eb7b822ecc056085ce8c035e
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0

qingfengxia, 2017-06-19:
thanks, i will look into that asap

Worufu3D, 2017-06-19:
Oh, so sorry, I noticed that this bug exists in jaheyns fork, I haven't check if this persists in this version.

closed 2020-01-22 by qingfengxia