qingfengxia/Cfd #21: Problem installing Cfd from freecad's 0.19 addons manager

aeropatron, 2020-09-16:
I am using linux ubuntuDDE 20.04, the appimage 22411(Git).

From freecad:
"Se han encontrado algunos errores que impiden instalar este entorno de trabajo: Falta módulo python: PyFoam, . Por favor, instale los componentes que faltan primero."

I translate from spanish:

Some errors has being found that prevent install this workbech, Misswed module python:PyFoam, Please install the missed components first.

From O.S. terminal:
:~$ sudo pip3 install PyFoam

Requirement already satisfied: PyFoam in /usr/local/lib/python3.8/dist-packages/PyFoam-2020.5-py3.8.egg (2020.5)
Requirement already satisfied: numpy in /usr/local/lib/python3.8/dist-packages/numpy-1.19.2-py3.8-linux-x86_64.egg (from PyFoam) (1.19.2)

qingfengxia, 2020-09-17:
Thank you, bug confirmed, also in Windows 0.19.  I will investigate it

qingfengxia, 2020-09-17:
I think it is due to a bug in PyFoam, although I think that is only limited to windows, seems not only.  
I have a quick fix: to make PyFoam importable

try comment out the line, and try import PyFoam in FC console and let me know : 
packages\PyFoam\Infrastructure\Configuration.py", line 85, 
    "descriptionPath": eval('["'+path.curdir+'","'+path.join(userDirectory(),"caseBuilderDescriptions")+'","'+path.join(globalDirectory(),"caseBuilderDescriptions")+'"]'),

here is my trial
>>> import PyFoam
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "D:\Software\FreeCAD_0.19.22411-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\PyFoam\__init__.py", line 7, in <module>
    from PyFoam.Infrastructure.Configuration import Configuration
  File "D:\Software\FreeCAD_0.19.22411-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\PyFoam\Infrastructure\Configuration.py", line 85, in <module>
    "descriptionPath": eval('["'+path.curdir+'","'+path.join(userDirectory(),"caseBuilderDescriptions")+'","'+path.join(globalDirectory(),"caseBuilderDescriptions")+'"]'),
  File "<string>", line 1
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape

qingfengxia, 2020-09-17:
I think your problem maybe,  PyFoam should be installed into the python inside appimage, not to system python.
I will update readme now

qingfengxia, 2020-09-17:
I will comment out PyFoam line in meta.txt if  PyFoam will not be fixed timely

aeropatron, 2020-09-17:
Hmm , from linux terminal:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named PyFoam

From python freecad terminal console, almost same result.
谢谢 Thanks for your work and help qingfengxia, Is there any web where I can follow a step by step guide on how to install the Cfd workbench now?
Others I tried from freecad's workbench manager had being installed fine and directly.

qingfengxia, 2020-09-18:
actually, there is a solution, please run the following lines in your FreeCAD terminal , assumming python3
`
import subprocess
import sys
print(sys.executable)
subprocess.check_output([sys.executable, "-m", "pip", "install", "PyFoam"])
# this should install PyFoam to the correct Python site, then try
import PyFoam
`

My Cfd may not working properly with latest 0.19. gmsh may not work. I have very limited time to maintain. 
you could always uninstall Cfd and install the fork CfdOF

aeropatron, 2020-09-18:
Hi again!

After install the requisites to Ubuntu of:
https://github.com/qingfengxia/Cfd

The workbench is installed!

But now at start I have this:
14:54:51  During initialization the error "module 'CfdTools' has no attribute 'getModulePath'" occurred in /home/myuser/.FreeCAD/Mod/Cfd/InitGui.py
14:54:51  Please look into the log file for further information

qingfengxia, 2020-09-19:
I think you still need to install PyFoam somehow.  
If you are on Ubuntu, you should use the PPA to install FreeCAD, then you can install PyFOAM to system python. 

I am sorry, AppImage is not supported.

closed 2020-09-23 by qingfengxia

shizukann, 2022-08-25:
> actually, there is a solution, please run the following lines in your FreeCAD terminal , assumming python3
> 
> ```python
> import subprocess
> import sys
> print(sys.executable)
> subprocess.check_output([sys.executable, "-m", "pip", "install", "PyFoam"])
> # this should install PyFoam to the correct Python site, then try
> import PyFoam
> ```
> 
> My Cfd may not working properly with latest 0.19. gmsh may not work. I have very limited time to maintain. you could always uninstall Cfd and install the fork CfdOF

Hi @qingfengxia 

Thank you for your instructions, however, I still cannot import PyFoam though it should be properly installed... 

```
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes 
in position 2-3: truncated \UXXXXXXXX escape
>>> import subprocess
>>> import sys
>>> print(sys.executable)
C:\Users\cuij\flow-env\Scripts\python.exe
>>> subprocess.check_output([sys.executable, "-m", "pip", "install", "PyFoam"])
b'Requirement already satisfied: PyFoam in c:\\users\\cuij\\flow-env\\lib\\site-packages (2021.6)\r\nRequirement already satisfied: numpy in c:\\users\\cuij\\flow-env\\lib\\site-packages (from PyFoam) (1.23.1)\r\n'
>>> import PyFoam
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\cuij\flow-env\lib\site-packages\PyFoam\__init__.py", 
line 7, in <module>
    from PyFoam.Infrastructure.Configuration import Configuration     
  File "C:\Users\cuij\flow-env\lib\site-packages\PyFoam\Infrastructure\Configuration.py", line 85, in <module>
    "descriptionPath": eval('["'+path.curdir+'","'+path.join(userDirectory(),"caseBuilderDescriptions")+'","'+path.join(globalDirectory(),"caseBuilderDescriptions")+'"]'),
  File "<string>", line 1
    [".","C:\Users\cuij\.pyFoam\caseBuilderDescriptions","/etc\pyFoam\caseBuilderDescriptions"]
                                                        ^
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes 
in position 2-3: truncated \UXXXXXXXX escape
>>>
```

Do you have any idea where went wrong with my installation? Thank you in advance!