qingfengxia/Cfd #24: Loading issue - Syntac error

mdziczkowski, 2021-07-16:
While attempting to load Cfd, I got following error message:

`During initialization the error "Command '['bash', '-i', '-c', 'source ~/.bashrc && echo $WM_PROJECT_VERSION']' returned non-zero exit status 1." occurred in share/Mod/Cfd/InitGui.py`

Note: For the security reason, the path was truncated

### System:

* System: Debian 10 (64-bit) - Linux
* FreeCAD version: 4.0.0

tultalk1, 2023-03-09:
18:53:40  b'your 131072x1 screen size is bogus. expect trouble\n'
18:53:40  environment var 'WM_PROJECT_VERSION' is not defined, fallback to hard-coded (5, 0)
18:53:41  During initialization the error "(unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape (<string>, line 1)" occurred in C:\Users\15172\AppData\Roaming\FreeCAD\Mod\Cfd\InitGui.py

qingfengxia, 2023-03-09:
FreeCAD version:  should not be 4.0.           what is your OpewnFOaM version? and is OpenFOAM under non-unicode path?